bengtmartensson/AGirs #7: Arduino-ifi code

probonopd, 2015-07-26:
Using the basic anatomy of an Arduino sketch

```
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

will help this to get widespread adoption in the Arduino community, as will using the [basic anatomy of an Arduino library](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification).

bengtmartensson, 2015-07-26:
Do not understand. Application codes has setup and loop, however not library code. Please explain.

probonopd, 2015-07-26:
Everything that ends in `.ino`is considered a "sketch" in the Arduino world and is supposed to have `setup()` and `loop()`.

bengtmartensson, 2015-07-26:
ok. The Arduino IDE also invokes a very badly designed, and on https://www.arduino.cc/en/Reference/HomePage completely undocumented,  preprocessor on .ino-files. This does not understand the C preprocessor, and thus makes a complete mess of files containing declarations withing #ifdef's, (The basic idea seams to generate forward declarations, just as if forward declaration was a good thing. It is not. They were invented to make one-pass compilers possible, nothing else, and not present in newer languages like Java.)  Also the IDE tends to fumble with it when it (!= me) considers it necessary. For this reason, I consider ino-files as unusable, write just sufficiently much to placify the IDE -- and I put a comment in them stating this fact. The real code goes into a .cpp-file.

The Arduino IDE is really a disaster.

probonopd, 2015-07-26:
As much as I respect your reasoning, please keep in mind that Arduino tries to separate all the "hard" (c++) stuff into libraries (written by developers) whereas normal users just use the simplified Arduino subset of the C language. 

While given your skills you might not see much value in this, please consider that your libraries might be used by Arduino users who do not understand their inner workings. This type of user will be entirely lost when the sketch is empty, and .cpp files deal with stuff (classes!) they have not been exposed to before.

bengtmartensson, 2015-07-27:
Please do the following: Rename Girs,cpp to Girs.ino (overwriting the latter),   (Note that now it is "complliant" with all Arduino rules I am aware of.) Try to build. You will see that the previously compiling file does not compile any longer. Moreover, the error messages are completely gibberish; referring to empty lines, and simply not understandable until you understand that they were cause by a "secret", undocumented preprocessor. Extremely hard for the expert to understand (It took me possibly hours to find out that there were a preprocessor. and what its problems were), not documented anywhere at arduino.cc (?). 

The problem is simply that the Arduino IDE is a very bad design, and that does not work for me. The problem is rather the interaction between the "secret" preprocessor and the C preprocessor that the usage of classes. 

I is/was planning to abandon the Arduino IDE cmpletely...

> While given your skills you might not see much value in this, please consider that your libraries 
> might be used by Arduino users who do not understand their inner workings.

Defending something that is genuinely bad by this sort of argument is simply dangerous!

probonopd, 2015-07-27:
That view is valid if you expect the contents of an .ino file to be C++. However most Arduino users might think about it as "Arduino language", a very limited subset. I don't think there is anything "secret" about the Arduino IDE, as it is apparently Open Source...

bengtmartensson, 2015-07-27:
Your view may be valid if you can point me to a place on arduino.cc where the "very limited subset" is unambiguously defined. In particular, where does Girs,cpp, renamed as Girs.ino, violate that subset? 

I did not write 'secret', I wrote '"secret"', meaning nowhere mentioned on arduino.cc.  

I strongly object to "user friendlyness" as excuse for doing things like invoking undocumented preprocessos behind the back of the user.

probonopd, 2015-07-29:
Perhaps you will enjoy reading http://hackaday.com/2015/07/28/embed-with-elliot-there-is-no-arduino-language/ which makes some points similar to yours. 

Still I would like to point out that typical "consumers" of Arduino libraries often do not really know about C++, so probably everything that needs a `->` is not meant to be in a .ino sketch (but instead abstracted away in a library).

Here is some pseudo code that feels more like an Arduino sketch to me:

```
# include (W)LAN libraries

LANInstance = LAN()
GirsInstance = Girs();

void setup() {
  Serial.begin();
  GirsInstance.ReceiveOnPin(2);
}

void loop() {
  if(GirsInstance.Received) {
      Serial.println(GirsInstance.Whateverwasreceived);
      LANInstance.send(GirsInstance.Whateverwasreceived);
      }
}
```

bengtmartensson, 2015-08-02:
I found this reference: http://www.gammon.com.au/forum/?id=12625 which explains the issues. Interestingly, I independently came up with the same solution the author recommends.

To your last example: This is not legal C++. C++ references are not like Java referenes, you can e.g. not assign to them. So (unfortunately) you cannot do very much objects in C++ without pointers.

Thank you for rising the issue. However, I think that further discussion should rather take place somewhere else, in an Arduino forum. 

Closing.

closed 2015-08-02 by bengtmartensson

probonopd, 2015-08-02:
Thanks for your response. I couldn't entirely figure out what is invalid in my pseudo-code example; however to make my point clear here is a real, working sketch that illustrates my point from above: https://github.com/marcoschwartz/aREST/blob/master/examples/ESP8266/ESP8266.ino