bengtmartensson/AGirs #15: Implementet IRSENSOR_X_PIN_PULLUP X = 1,2,...

bengtmartensson, 2016-01-24:
Applies to both AGirs and GrisLite, (possibly Girs4Lirc). Cf https://github.com/bengtmartensson/harctoolboxbundle/issues/92#issuecomment-174245232

For this, there should be a second argument of IrWidgetAggregating::newIrWidgetAggregating in capture(Stream&).

BartmanEH, 2016-01-31:
As a work-around, can I simply add the following to *GirsUtils.h*:
```
#ifdef IRSENSOR_1_PIN
        pinMode(IRSENSOR_1_PIN, INPUT_PULLUP);
#endif
```
I know this isn't a scalable solution but rather a hack. Will it work? I can't tell if it will conflict with code elsewhere in the project.

bengtmartensson, 2016-01-31:
No it will not work. Try the commit instead.

BartmanEH, 2016-01-31:
```
sketch\GirsLite.cpp: In function 'boolean capture(Stream&)':
GirsLite.cpp:293: error: 'sensorPullup' is not a member of 'GirsUtils'
             GirsUtils::sensorPullup(sensorNo));
```

bengtmartensson, 2016-01-31:
There must be something "stale" somewhere, like an old version of GirsUtils.h, likely in the library.

BartmanEH, 2016-01-31:
I believe you were correct. I uninstalled the IDE, deleted the AppData folder manually, rebooted windows as suggested [here](https://github.com/arduino/Arduino/issues/4137#issuecomment-158629254), reinstalled the IDE, recopied the libraries *GirsLib*, *Infrared4Arduino* and *LiquidCrystal_I2C*, and unbelievably I was able to compile it! Testing will be tomorrow evening once I get my Pro Micro and remote controls back into my hands.

BartmanEH, 2016-01-31:
This is awesome. I poked around the newly committed source code to learn about the new *IRSENSOR_X_PULLUP X = 1,2,...* implementation. I want to thank Bengt for his amazing work here - going beyond his own needs and adding a lot of new code to support alternative hardware like mine. Now I can just go into the pin definition file *girs_pins_micro.h* for my Pro Micro or *girs_pins_default.h* for my Uno R3 Plus and just uncomment the pullup line here:
```
#if defined(CAPTURE)
// capture pin (ICP) 4
#define IRSENSOR_1_PULLUP
```
...and have the AtMega chip do the hard work of pulling up my open collector output on my QSE159 IR sensor rather than having to sneak in a resistor in the hardware. Thanks so much!

BartmanEH, 2016-02-01:
The latest committed AGirs is missing the ability to detect and use *girs_pins_micro.h* since the *config.h* file has:
```
// Include one file describing the pin configuration
#ifdef ARDUINO_AVR_NANO
//#include <girs_pins_nano_shield.h>
#include <girs_pins_nano.h>
#else
#include <girs_pins.h> // Generic
#endif
````

BartmanEH, 2016-02-01:
Nevermind... I see that in *girs_pins.h* there are further #elif directives:
```
#if defined(ARDUINO_AVR_NANO)
#include "girs_pins_nano.h"
#elif defined(ARDUINO_AVR_MICRO)
#include "girs_pins_micro.h"
[...]
```

bengtmartensson, 2016-02-01:
You have missed the semantics of the line
`// Include one file describing the pin configuration`

It means that the user is supposed to add/adapt a/the include statements for a/the  desired configuration file. Of course, if maintaining several configurations, using preprocessor symbols may be handy, if the user so desires. My aim is not to have a ladder of interleaved `#ifdef` and `#includes`.  So I do not consider it a mistake.

Of course, suggestions for improvements are welcome.

Glad that the pullup works for you, closing.

closed 2016-02-01 by bengtmartensson

BartmanEH, 2016-02-01:
There is already a ladder of interleaved *#ifdef* and *#includes* in *girs_pins.h* - what's confusing is that it is in two different places. Perhaps the ladder from *girs_pins.h* could be moved to the corresponding section in *config.h*

bengtmartensson, 2016-02-02:
Thank you for the suggestion. I like to emphasize  "User, add the file YOU want" instead,

> The verbose output of a compile with the Arduino IDE doesn't show the usage of either 
> girs_pins_default.h or any other girs_pins_[board].h file so I'm not sure how to prove this.

If I want to check that a certain file is included, i just put

`#error blahblah`

into it and check for an error `blahblah' when compiling.

BartmanEH, 2016-02-02:
*#error blahblah* is a great tip, thanks for that. I've put it in my personal Cheat Notes file.