bengtmartensson/AGirs #16: GirsLite using Arduino Pro Micro

BartmanEH, 2016-01-26:
Thank you for producing a *girs_pins_micro.h* file to support the Pro Micro. I take it from the following section:
```
#ifdef LED
#define SIGNAL_LED_1 2
//#define SIGNAL_LED_1_GND 3
#define SIGNAL_LED_4 A0
#define SIGNAL_LED_4_GND 15
#define SIGNAL_LED_3 A2
#define SIGNAL_LED_3_GND A1
#define SIGNAL_LED_2 A3
//#define SIGNAL_LED_1_GND 8
#endif
```
that I would wire up the LEDs as follows:

| LED | colour | cathode | anode |
|---------|----------|----------|----------|
| LED 2 | Red | A3 | ground |
| LED 3 | Yellow | A2 | A1 |
 LED 4 | Green | A0 | pin 15 |

1. Is the LED wiring immediately above correct?
2. What about the fact the [Pro Micro has no built in LED](http://cdn.sparkfun.com/datasheets/Dev/Arduino/Boards/Pro_Micro_v13b.pdf)? What does `#define SIGNAL_LED_1 2` end up doing for Pro Micro?
3. To direct GirsLite to use the new Pro Micro pin definition file, I think I need to add `#define ARDUINO_AVR_MICRO` to *GirsLite.ino*, correct?

bengtmartensson, 2016-01-30:
The existence of said file does not mean that I "support" something; I did something sometime ago, possibly it works for others too... The file corresponds to the hardware in the photo 

[image]

which should answer the pin stuff.

I put down a lot of effort into (the Nano instruction)[http://www.harctoolbox.org/arduino_nano.html], using other HW is in this light a much harder task.

2. No problem. GirsLite does not use LED1, except for the using the command "led 1 on" (on the "command line") and the self test. On my picture, there is a blue LED1.

3. Not correct. The Arduino IDE defines a preprocessor symbol, e.g. ARDUINO_AVR_NANO, accroding to the board you have selected Tools -> Board.

BartmanEH, 2016-01-31:
I understand now and I can now make the correlation of Arduino Nano pins and translate them to Arduino Pro Micro pins. Now I know about the preprocessor symbols too - thanks for that. The Arduino IDE really hides too many things.

I reviewed the entire Nano hardware guide and notwithstanding my challenges to switch to the Pro Micro, information is missing about how to change the I/O pin assignments for different non-demodulating IR sensors. The QSE159 I have has a different pin out and the pin definitions need to be changed to accommodate the VCC/signal/GND pinout. So after reading through many of your source files, I see the following section in *girs_pins_micro.h*:
```
#ifdef RECEIVE
#define IRRECEIVER_PIN 10
#define IRRECEIVER_GND 16
#define IRRECEIVER_VSS 14
#endif
```
and I see *GirsUtils.h* where the digital levels are set for those I/O pins. What's confusing is that Vss is usually used for negative supply or ground, however, I think you intend for IRRECEIVER_VSS to be the positive supply for the IR sensor, correct?

bengtmartensson, 2016-01-31:
You are right, there is no beginners guide to using Micro/QSE159 etc. I have no immediate plans to write one either, but if someone sends me a useful text, I will consider using or incorporating it. I have written one beginners guide already....

BartmanEH, 2016-01-31:
Suggest you search & replace _VSS for _VCC in your pin definition source code files. I will proceed assuming this is correct.

BartmanEH, 2016-02-01:
I think there's a small bug in *girs_pins_micro.h*. The following:
```
#if defined(CAPTURE)
// capture pin (ICP) 4
//#define IRSENSOR_1_PULLUP
#define SENSOR_GND 5
#define SENSOR_VSS 6
#endif
```
should be:
```
#if defined(CAPTURE)
// capture pin (ICP) 4
//#define IRSENSOR_1_PULLUP
#define IRSENSOR_1_GND 5
#define IRSENSOR_1_VSS 6
#endif
```
The parameters *SENSOR_GND* and *SENSOR_VSS* are not referenced in *GirsUtils.h* - they are referenced as *IRSENSOR_1_GND* and *IRSENSOR_1_VSS*.

bengtmartensson, 2016-02-01:
Thanx!

bengtmartensson, 2016-02-03:
Everything appears to have been cleared out, closing.

closed 2016-02-03 by bengtmartensson