bengtmartensson/AGirs #19: Arduino Error Compiling

janvikalra, 2016-07-14:
The Girs.Lite.cpp in the sketch is blank which leads to the following error message: 

Arduino: 1.6.6 (Mac OS X), Board: "Arduino/Genuino Uno"
stat /Users/janvikalra/Documents/Arduino/libraries/GirsLite/GirsLite.cpp: no such file or directory
Error compiling.

I tried the same sketch on a windows and got the same error message.

bengtmartensson, 2016-07-14:
No, it isn't. Something must have gone wrong when you checked out it. GirsLite.cpp is a symbolic link to ../Girs/Girs.cpp:

    $ pwd                            
    /home/bengt/harctoolbox/AGirs/src/GirsLite
    $ ls -l GirsLite.cpp 
    lrwxrwxrwx. 1 bengt bengt 16 Jan  2  2016 GirsLite.cpp -> ../Girs/Girs.cpp

Possibly you git mishandles symbolic links?

As a list resort, replace  GirsLite.cpp  with the content of Girs.cpp.

janvikalra, 2016-07-14:
Hmm, maybe I did something wrong in the process then that caused mishandling? Here's what I did: 
1. Download Zip file of the master from Github 
2. Unzip file
3. Drag the GirsLite folder into Arduino > Library 
4. Open Arduino and open the GirsLite file by File> Examples > GirsLite
5. Upload the Sketch 
Do you see anything wrong in the process?

bengtmartensson, 2016-07-14:
I discovered that already in the zip file (generated by Github by their software) GirsLite.cpp is broken. Sigh... 

The recent commit 95f0c77 replaces that symlink by a copy. Try "Version 0.2.0".

Hope it works now for you. Thank you very much for bringing this problem to my attention.

janvikalra, 2016-07-14:
No worries - thanks for your quick response! 

So I download Version 0.2.0 as you suggested but unfortunately it still doesn't seem to compile on my Arduino. 
[image]

These is what the error message says:

> Arduino: 1.6.6 (Mac OS X), Board: "Arduino/Genuino Uno"
> 
> In file included from /Users/janvikalra/Documents/Arduino/libraries/GirsLite/GirsLite.ino:5:0:
> sketch/config.h:58:34: fatal error: girs_pins.h: No such file or directory
>  #include <girs_pins.h> // Generic
>                                   ^
> compilation terminated.
> exit status 1
> Error compiling.

I then tried to comment out the " #include <girs_pins.h> // Generic" line but then this error came up: 

> Arduino: 1.6.6 (Mac OS X), Board: "Arduino/Genuino Uno"
> 
> /var/folders/4j/5zcy_kmj4zn2q13wppt2j_300000gn/T/arduino_314c61afcf175aabbd385bb5f5d6ea8d/GirsLite.ino:7:27: fatal error: InfraredTypes.h: No such file or directory
>  #include <InfraredTypes.h>
>                            ^
> compilation terminated.
> exit status 1
> Error compiling.

Could it be that I am uploading the sketch incorrectly? I used the same process I describe in my last post.

bengtmartensson, 2016-07-14:
Problem 1.: Something seriously wrong here. There is no `#include // Generic` line anywhere! GirsLite.ino looks like 

    // Dummy file to placify the silly Arduino IDE.
    // The real program resides in Girs.cpp.
    // Rationale: see http://www.gammon.com.au/forum/?id=12625

    #include "config.h"

    #include <InfraredTypes.h>
    #include <GirsUtils.h>

can you investigate? 

Problem 2. You have not installed Infrared4Arduino (correctly). It can be installed from the library manager in the Arduino IDE.

janvikalra, 2016-07-14:
Oh gosh, there is a mistake in my question above; the error message actually said: 

> Arduino: 1.6.6 (Mac OS X), Board: "Arduino/Genuino Uno"
> 
>  file included from /Users/janvikalra/Documents/Arduino/libraries/GirsLite/GirsLite.ino:5:0:
> sketch/config.h:58:34: fatal error: girs_pins.h: No such file or directory
>  #include **girs_pins.h** // Generic

Sorry about that! The #include girs_pin.h is in the **config.h** tab of the GirsLite sketch for me. 

Although I had not previously downloaded Infrared library via the library manager, I just did and the same errors still prevent me from compiling. Does this version compile correctly on your computer? 

_Also for the record, it's amazing that you have made your own official library for Arduino and I am actually talking to you right now. WOW!_

bengtmartensson, 2016-07-14:
Try to update "Arduino" to 1.6.9, I think it will help. (girs_pins.i is in ../GirsLib.)

The libraries the Arduino library manager knows about are not "official Arduino libraries".

janvikalra, 2016-07-17:
Hii, unfortunately, I still couldn't get it to work even after updating my Arduino IDE - I think there's something wrong on my end. I will be using another software to get the same information. Thanks for all your help, regardless! Much appreciated!

closed 2016-07-17 by janvikalra

bengtmartensson, 2016-07-17:
If I may:

First make sure that building simple examples work. (Support by the Arduino forum.) Then start from scratch, follow the instructions _exactly_, without being "creative". It that does not work, either the instructions are wrong/incomplete, or there bug. I like to know that.