bengtmartensson/AGirs #22: Support for ESP8266

klattimer, 2017-01-20:
I'm planning on building this board http://fritzing.org/projects/agirs-dht-non_mod-433-tx that I've designed over the weekend and I'm hoping to get around to building support for ESP8266 via the library - I *think* this will mostly be a swap in over the ethernet library which I believe is much the same.

bengtmartensson, 2017-01-20:
Cool project! The hardware dependencies are all in the library [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino); there is already a [ticket for porting that](https://github.com/bengtmartensson/Infrared4Arduino/issues/5). Good luck!

bengtmartensson, 2017-03-19:
Closing, since this is covered by [this ticket](https://github.com/bengtmartensson/Infrared4Arduino/issues/5).

closed 2017-03-19 by bengtmartensson