bengtmartensson/AGirs #35: Wrong #ifndef in file girs_pins_nano_shield.h

gklement, 2017-11-17:
Drop file from repos or change
#ifndef GIRS_PINS_NANO_H
#define GIRS_PINS_NANO_H
into
#ifndef GIRS_PINS_NANO_SHIELD_H
#define GIRS_PINS_NANO_SHIELD_H

but values will still depend on include order.

bengtmartensson, 2017-11-19:
Thank you very much for the observation. Actually, these files (`girs_pins_*.h`) are not headers in the strict sense, so include guards in the traditional sense do not really make sense here, although I agree that it is currently quite confusing. I have instead opened the more general issue #36, that addresses this and other problems. Your feedback and other contributions are most welcome there.

closed 2017-12-07 by bengtmartensson

bengtmartensson, 2017-12-07:
This project is now using #pragma once instead of the legacy include guards.

commit 47bd7f9a (2017-12-07): Misc. fixes to configuration. Resolves #35.