bengtmartensson/AGirs #38: issuing a "send" immediately after a "receive" requires waiting for the receive to complete or time out

ebukh, 2018-06-01:
If I issue a send command after a receive command to GirsLite, I end up waiting for the receive to either get a signal, or to time out (default is 10 seconds) before my send command is executed. 
I would like to have a way to interrupt the receive in GirsLite before the timeout elapses, by sending a command such as "end receive", and having it take effect immediately.

bengtmartensson, 2018-06-01:
The "receive" command terminates immediately if there are data available in the input stream. So (for example) sending a blank line achieves what you are asking for. You can try it manually in the Arduino serial monitor.

The "capture" command is not incorruptible though.

ebukh, 2018-06-01:
I tired it over serial, Issued a receive command (receive led goes on). Then immediately pressed enter (i.e. send a blank line). It took ~10s to get back the OK corresponding to the blank line. I got the receive timeout response (".") first.

bengtmartensson, 2018-06-01:
Current version? Try setting the line ending (of the Arduino serial console) to "Carrage return" (bottom row, first entry of the right part).

ebukh, 2018-06-01:
Yes ( I think). Set it to carriage return (I had it set to \r\n, which is what my program does), but it still does the same behavior.

bengtmartensson, 2018-06-01:
I cannot reproduce; works perfectly for me. Please describe the stuff you are using, your operating system, board, version of the Arduino IDE, version of the AGirs, exactly how you installed etc.

You can also try to send another string like "sdkfsd" to interrupt.

ebukh, 2018-06-01:
My OS is windows 7 x64. Board is Arduino nano chinese clone (atmega 328p). USB driver is whatever is installed by the OS automatically. My guess is arduino IDE version doesn't matter, since I can reproduce the same behavior in my own code, but I have version 1.8.5. The firmware was downloaded from here: https://github.com/bengtmartensson/AGirs/files/76248/GirsLite.nano.hex.zip, I also tried compiling my own version, with similar results.

bengtmartensson, 2018-06-01:
Ok, problem solved. That version is many releases old, 2 and a half years. Yes, it shows the behavior you do not like. Update to the [current release 1.0.0.](https://github.com/bengtmartensson/AGirs/releases) alternatively install AGirs from the Arduino library manager; it is example "Girs". (Must install "Infrared" too).

ebukh, 2018-06-01:
That works, thanks a lot. You should update the firmware-related download links in your hardware [guide](http://www.harctoolbox.org/arduino_nano.html).

bengtmartensson, 2018-06-02:
Glad that it is resolved. Ok to close?

Feel free to open issues for the other stuff you mentioned in the personal mail.

closed 2018-06-02 by ebukh