bengtmartensson/AGirs #52: Wrong capture data on Nano, Girs 1.0.2 and IR sensor from 37 in 1 sensors kit (TSMP58000?)

joric, 2020-04-14:
Can't capture proper codes, I'm getting repeated values for all the buttons like:

`+10,-1694,+10,-1694,+10,-1694,+10,-1694,+10,-1694,+10,-1694,+10,-1694,+10,-1694,+10,-1694,+10,-1694,+10,-1694,+10,-8428`

It should be a real signal with variable content, e.g. (RAW format, grabbed with IRController.ino on ESP8266):
`1330,438,1258,440,410,1288,1260,438,1258,440,384,1312,386,1312,412,1286,386,1312,386,1312,1260,440,384,8106`

I'm using IR sensor with metal shielding from 37 in 1 kit and Arduino Nano. This sensor only works if connected with signal pin to D8, doesn't work on D5 so I'm assuming it's (TSMP58000). Also it uses separate power pins, can't be powered from GPIO pins, so I'm connecting power pins directly to 5V and GND. Using 3V pin instead of 5V doesn't help.

Any chance I could make it work? I don't have other sensors. I'm using Windows 10.

Also see https://github.com/joric/irsend for pictures.

bengtmartensson, 2020-04-14:
It is almost certainly a demodulating receiver, possibly VS1838. 

So define IRRECEIVER_1_PIN to 8, undefine CAPTURE, and use "receive" instead of "capture", and you should be set.

joric, 2020-04-14:
Thanks a lot! Though it wasn't necessary to recompile, I ended using a standard 1.0.2 hex with receiver hooked up to D5, I've just checked "Use receive for capture" in the IrScrutinizer UI and it worked! Build instructions could be clearer though :) Who knew that Arduino firmware is in the examples folder and it's called Girs.ino and the ino file is supposed to be empty and the dependence library in the library manager shoud've been called precisely "infrared", not Girs. Maybe the whole building section needs rewriting. Anyway, thanks a lot, I've managed to capture the signals now.

closed 2020-04-14 by joric

bengtmartensson, 2020-04-14:
Glad that it works.

I have put quite some effort into packaging and documentation, however if you can suggest improvements, preferably as PR or patches, it is most welcome.

There is a library manager but not "sketch manager", explaining the "example". The reason for the almost-empty Girs.ino is given as comment in that file (I consider the Arduino pre-processor to be a *very bad idea*, causing more problems than it solves). The low level library is automatically loaded by the library manager.