bengtmartensson/AGirs #53: error compiling GirsLite

newmguy, 2020-06-09:
I have read the other posts regarding this same error but I did not find a solution in those posts.
Here is the error:
`Arduino: 1.8.11 (Windows 7), Board: "Arduino Nano, ATmega328P"

In file included from C:\Users\user\Documents\Arduino\GirsLite\GirsLite.ino:5:0:

config.h:59:10: error: girs_pins_nano.h: No such file or directory

 #include <girs_pins_nano.h>

          ^~~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1
girs_pins_nano.h: No such file or directory
`
I copied the file girs_pins_nano.h to the local directory and it still gives the error.
Any help would be greatly appreciated.

bengtmartensson, 2020-06-10:
I cannot reproduce.
Please delete current installation and install through the Arduino library manager.

bengtmartensson, 2020-06-13:
No feedback received. Since three days has passed, and I am pretty sure that your problem is not a flaw in my software, I close. Re-open if you have a different opinion.

closed 2020-06-13 by bengtmartensson