bengtmartensson/AGirs #57: Continuous receive to detect if NEC1/2 sends dittos or repeats for long key presse.

gkle, 2021-02-14:
I am using AGirs and IrCrutinizer for years to add new devices to my URC1280 remote (finally ended up to this model).
But decoded NEC signal has no hint if long key press sends dittos or repeats. At least i didn't figure it out.
So i extended the commands of AGris within processCommand with this lines of code to constinuously decode the signal.
    if (cmd[0] == 'e') {
        while(!stream.available()) {
            receive(stream);
        }
    } else
Pressing any key will exit the loop.
This works only within a seriell terminal pressing e+Enter.

PS: Did not figure out what the tokenizer is for. Skipping it at all. I've read the other issue thinking it's a good idea to kick it off.

gkle, 2021-02-14:
PPS: May also apply to detect macros.

closed 2021-02-14 by gkle

bengtmartensson, 2021-02-14:
Are you aware of the [parameter receiveendingtimeout](https://github.com/bengtmartensson/AGirs#parameters)? By setting this to (say) 100 it will take an 100 ms of silence to consider the signal as ended, which means that for example a NEC* signal will not be cut between the dittos. This is, I think, what you desire.