bengtmartensson/AGirs #71: AGirs/examples/Girs/ is incomplete

Dan-Essin, 2023-08-16:
Girs.ino says:
#include "config.h"
#include <InfraredTypes.h>

InfraredTypes.h is not provided therefore the code is useless as it stands. Other Agirs projects have a similar. There are references to many .h files that cannot be found anywhere on GitHub or anywhere else on the internet, for that matter.

I know that alot of folks have put effort into this but why not finish the job so that the projects and examples will actually be usable?

Dan-Essin, 2023-08-16:
I searched all of GitHub for InfraredTypes.h. These are the results:
**Your search did not match any repositories**
InfraredTypes.h appears in 55 code samples and 3 issues but the actual file does not exist. It would take a mind-reader to divine what it should contain.

bengtmartensson, 2023-08-16:
This is described in the documentation, which you -- obviously -- did not read.

I find your tone to be rude and offensive, and I would strongly suggest that you adapt a more civil language.

closed 2023-08-16 by bengtmartensson